angeloc/simplemodbusng #6: Examples for Slave are not working, missing lowLatency argument

Krosik, 2016-06-13:
Hi

in both Slave and SoftwareSerialSlave examples is error, because in
modbus_configure(115200, 1, 2, TOTAL_REGS_SIZE);

is mising last argument for

void modbus_configure(long baud, unsigned char _slaveID, unsigned char _TxEnablePin, unsigned int _holdingRegsSize, unsigned char _lowLatency)

Also this argument is not explained in examples or comments. I believe it should be 0 or 1, maybe better boolean true or false, but I do not understand what is exact use of it.

thx

0mit, 2016-06-15:
I think you should read the comments below the function implementation in the source file [HERE](https://github.com/angeloc/simplemodbusng/blob/master/SimpleModbusSlave/SimpleModbusSlave.cpp)
But you can use 0 or anything if you don't know what is it for and you won't.

Krosik, 2016-06-17:
ok but please edit example files accordingly, so people who do not understand much code will have no error during compilation.

thx

angeloc, 2016-06-27:
@Krosik 

Please, make a pull request with a fix, I'm eager to accept it!

Krosik, 2016-07-10:
ok done Angelo

closed 2016-07-10 by Krosik