angeloc/simplemodbusng #7: Feature request : Softserial for master

Memes11, 2016-06-17:
all in title, I am looking for that desperately

angeloc, 2016-06-27:
Hi @Memes11,

How much are you willing to pay?

Memes11, 2016-07-11:
Hi angeloc, maybe some of my time to help you test, since I am not desperate anymore, found another solution, turned out I did not need it :)

nkl-nklc, 2017-03-01:
Hi @Memes11,

could you sare with us how you implemented software serial ?

akshulg, 2017-12-11:
@Memes11 can you share the solution for software serial for modbus master?

akshulg, 2018-01-03:
@Memes11 @angeloc @angeloc is it okay to read the register values from slave using the rx and tx pin of arduino and at the same time printing those values on serial monitor. I am implementing arduino as master and raspberry pi as slave but getting garbage values on serial monitor. Any idea why is this happening?

limitedAtonement, 2020-08-09:
@akshulg I think your question has nothing to do with this issue. To answer your question, that should be fine as long as you have multiple UART modules. Perhaps you are using the same serial module for monitoring and RS-485?

closed 2020-11-10 by angeloc