angeloc/simplemodbusng #10: Use the Stream class to allow other Serial interfaces

elsp1991, 2016-08-03:
Iam using this library with arduino micro and I hace modified it in order to use the Serial1 instead of the Serial (emulated USB). It would be nice if we can select the serial in the constructor using the Stream class

brainelectronics, 2019-07-15:
Ceckout the latest merge. I've added SoftwareSerial as config parameter. I can also add Hardware Serial if you wish

matthijskooijman, 2019-09-05:
> Ceckout the latest merge. I've added SoftwareSerial as config parameter. I can also add Hardware Serial if you wish

Better accept a Stream* or Stream& and work on that. Stream is the superclass of Both HardwareSerial and SoftwareSerial (but also AltSoftSerial and other things). The only thing Stream does not offer is baudrate control through a `begin()` function, so when passing a Stream to this library, the baudrate should be configured by the caller beforehand.

elsp1991, 2019-09-06:
That sounds great exactly what I had in mind
 thanks

closed 2020-11-10 by angeloc