angeloc/simplemodbusng #12: how to read slave registers from master..?? please kindly help..any answer is welcome.. im open to suggestion..

laxmisundar, 2017-05-06:


akshulg, 2017-12-11:
@laxmisundar  were you able to find the solution? I am working on this currently. Modbus protocol between Arduino(master) and Atmega16(slave).

laxmisundar, 2017-12-12:
Yes i was succesfull.. @akshulg

akshulg, 2017-12-30:
@laxmisundar what was the solution? please tell me

laxmisundar, 2017-12-30:
At slave example program u would hv seen some codes like digitalRead[holdingregister[i]]... those are the registers which has been transmitted to your master registers.. at ur master arduino if u Serial print the reading registers u would  come to know tht..

akshulg, 2017-12-30:
I am actually working on modbus between raspberry pi and arduino. Successfully done pi as master and arduino as slave. Used the simplemodbus slave for arduino and pymodbus client example for pi. pymodbus is a python library available on github only. Now i wanted to set arduino as master and pi as slave. So using pymodbus server example for pi and simplemodbus master for arduino. But getting some error.

akshulg, 2017-12-30:
DEBUG:pymodbus.server.sync:Client Connected [/dev/ttyUSB0:/dev/ttyUSB0]
DEBUG:pymodbus.server.sync:Started thread to serve client
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0xa 0xc5 0xcd
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 0:10
DEBUG:pymodbus.datastore.context:getValues[3] 0:10
DEBUG:pymodbus.server.sync:send: 010314000a000a000a000a000a000a000a000a000a000a416d
DEBUG:pymodbus.server.sync:0x1 0x10 0x0 0x6 0x0 0x9 0x12 0x0 0xa 0x0 0xa 0x0 0xa 0x0 0x1 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd5 0x85
DEBUG:pymodbus.factory:Factory Request[16]
DEBUG:pymodbus.datastore.context:validate[16] 6:9
DEBUG:pymodbus.datastore.context:setValues[16] 6:9
DEBUG:pymodbus.server.sync:send: 011000060009e00e
DEBUG:pymodbus.server.sync:0xa 0x0 0x0 0x0 0x0 0xa 0x80 0xb6
DEBUG:pymodbus.factory:Factory Request[10]
ERROR:pymodbus.server.sync:Socket error occurred Modbus Error: [Invalid Message] <pymodbus.pdu.IllegalFunctionRequest object at 0x766f92f0>
DEBUG:pymodbus.server.sync:0x1 0x10 0x0 0x6 0x0 0x9 0x12 0x0 0xa 0x0 0xa 0x0 0xa 0x0 0x2 0x0 0x1 0x0 0x1 0x0 0x1 0x0 0x1 0x0 0x0 0xa6 0x1

akshulg, 2017-12-30:
@laxmisundar any idea, hint will be helpful

angeloc, 2020-11-10:
@akshulg You question is better suited to pymodbus support.

closed 2020-11-10 by angeloc