angeloc/simplemodbusng #19: Modbus RTU slave: auto baud rate detection

tuxmartin, 2018-08-08:
Hi,

is possible to use auto baud detection for Modbus RTU slave?

Something like this: https://gist.github.com/tuxmartin/46ca107057c650fbe6f63dcae5ee067d

My idea:

- Modbus Master (Raspberry Pi) set serial port speed to X.
- ? (Modbus Master send broadcast message for synchronization every X seconds.) ?
- Modbus Slave (Arduino) run detection of speed every X seconds.
- Modbus Master periodically read data from all slaves - if any of slaves do not respond, slow down serial speed.

Thanks.

limitedAtonement, 2020-08-09:
This can probably be closed, but if the question still stands, it depends on the medium over which modbus is taking place. Modbus doesn't specify the underlying protocol, and it is that protocol that would need to handle the baud detection. RS-485 doesn't have a good way to do baud rate detection. The signal is high for some amount of time to signal a message is about to come, then the line goes low for the start bit. Perhaps you could take a timing on the signal going from "medium" to high to low as one bit, and set the clock accordingly? I'm not sure if RS-485 allows that, and I'm not sure how badly that could go. The measurement would have to be quite accurate.
After the start bit, if, for instance, the number "0" is being transmitted, there is simply a low signal on the line the whole time until the stop bit or parity bits. There is no timing information.

Perhaps you should look at IIC which also is a two-wire protocol, but it's very different: one wire is a clock that goes up and down the whole time for every bit. Using this protocol, it is easy to detect the baud.

tuxmartin, 2020-08-12:
I would still like autodetection.

I use RS-485.

Hayes detection works on RS-232: https://en.wikipedia.org/wiki/Hayes_command_set#Autobaud

Is something similar possible on RS-485 with Moddbus RTU?

limitedAtonement, 2020-08-12:
To my knowledge, it's not a normal thing. You'll have to implement it yourself which won't be easy. Again, as I said, maybe IIC would be better for you as auto baud detection is trivial there because of the protocol.

closed 2020-11-10 by angeloc