angeloc/simplemodbusng #20: write to multiple registers help

heinpheiifer, 2018-11-17:


heinpheiifer, 2018-11-17:
Hi there i am can read value but not write values here is my code ,could anybody help me please .

#include <SimpleModbusMaster.h>

/*
   The example will use packet1 to read a register from address 0 (the adc ch0 value)
   from the arduino slave (id=1). It will then use this value to adjust the brightness
   of an led on pin 9 using PWM.
   It will then use packet2 to write a register (its own adc ch0 value) to address 1
   on the arduino slave (id=1) adjusting the brightness of an led on pin 9 using PWM.
*/

//////////////////// Port information ///////////////////
#define baud 4800
#define timeout 2000
#define polling 10 // the scan rate
#define retry_count 10

// used to toggle the receive/transmit pin on the driver
#define TxEnablePin 4

#define LED 9

// The total amount of available memory on the master to store data
#define TOTAL_NO_OF_REGISTERS 2

// This is the easiest way to create new packets
// Add as many as you want. TOTAL_NO_OF_PACKETS
// is automatically updated.
enum
{
  PACKET1,
  PACKET2,
  PACKET3,
  PACKET4,
  PACKET5,
  PACKET6,
  PACKET7,
  PACKET8,
  PACKET9,
  PACKET10,
  PACKET11,
  TOTAL_NO_OF_PACKETS // leave this last entry
};

// Create an array of Packets to be configured
//Packet packets[TOTAL_NO_OF_PACKETS];
Packet packets[TOTAL_NO_OF_PACKETS];

packetPointer motorVoltageVSD = &packets[PACKET1];
packetPointer actualSpeedVSD = &packets[PACKET2];
packetPointer motorCurrentVSD = &packets[PACKET3];
packetPointer motorPowerVSD = &packets[PACKET4];
packetPointer statusWordVSD = &packets[PACKET5];
packetPointer motorTimeVSD = &packets[PACKET6];
packetPointer motorFrequencyVSD = &packets[PACKET7];
packetPointer motorTemperatureVSD = &packets[PACKET8];

packetPointer commandWordVSD = &packets[PACKET9];
packetPointer userSetSpeedVSD = &packets[PACKET10];
packetPointer clearFaultsVSD = &packets[PACKET11];

////////VSD READ VARIABLES///////////
unsigned int readMotorVoltageVSD[1];
unsigned int readActualSpeedVSD[1];
unsigned int readMotorCurrentVSD[1];
unsigned int readMotorPowerVSD[1];
unsigned int readStatusWordVSD[1];
unsigned int readMotorTimeVSD[1];
unsigned int readMotorFrequencyVSD[1];
unsigned int readMotorTemperatureVSD[1];

////////VSD WRITE VARIABLES//////////
unsigned int writeControlWordVSD[1];
//unsigned int writeUserSetSpeedVSD[1]={0};
unsigned int writeClearFaultsVSD[1];

// High or Low variables as arrays (Not used)
unsigned int writeHigh[1]={1};
unsigned int writeLow[1]={0};

int StartButton = 2;
unsigned int BUTTONSTATE=0;

void setup()
{
  Serial.begin(4800);
  delay(100);
  pinMode(StartButton, INPUT);
  // Initialize each packet for reading
  modbus_construct(motorVoltageVSD, 1, READ_HOLDING_REGISTERS, 3207, 1, readMotorVoltageVSD);
  modbus_construct(actualSpeedVSD, 1, READ_HOLDING_REGISTERS, 8604, 1, readActualSpeedVSD);
  modbus_construct(motorCurrentVSD, 1, READ_HOLDING_REGISTERS, 3204, 1, readMotorCurrentVSD);
  modbus_construct(motorPowerVSD, 1, READ_HOLDING_REGISTERS, 3211, 1, readMotorPowerVSD);
  modbus_construct(statusWordVSD, 1, READ_HOLDING_REGISTERS, 8603, 1, readStatusWordVSD);
  modbus_construct(motorTimeVSD, 1, READ_HOLDING_REGISTERS, 3233, 1, readMotorTimeVSD);
  modbus_construct(motorFrequencyVSD, 1, READ_HOLDING_REGISTERS, 3202, 1, readMotorFrequencyVSD);
  modbus_construct(motorTemperatureVSD, 1, READ_HOLDING_REGISTERS, 3209, 1, readMotorTemperatureVSD);

  // Initialize each packet for writing

  modbus_construct(commandWordVSD, 1, PRESET_MULTIPLE_REGISTERS, 8601, 1,writeControlWordVSD);
 //modbus_construct(userSetSpeedVSD, 2, PRESET_MULTIPLE_REGISTERS, 8602, 1, readUserSetSpeedHMI);
  modbus_construct(clearFaultsVSD, 2, PRESET_MULTIPLE_REGISTERS, 8501, 1, writeClearFaultsVSD);
  // Initialize the Modbus Finite State Machine

  modbus_configure(&Serial, baud, SERIAL_8E1, timeout, polling, retry_count, TxEnablePin, packets, TOTAL_NO_OF_PACKETS);
  pinMode(LED, OUTPUT);
  delay(100);
}

void loop()
{
  modbus_update();
  delay(100);
  Serial.println ("ATV312 ARDUINO");
  int statusword = readStatusWordVSD[0];
  int voltage = readMotorVoltageVSD[0]/10;
  float motorCurrent = readMotorCurrentVSD[0];
  int current = readMotorCurrentVSD[0]/10;
  int frequency = readMotorFrequencyVSD[0]/10;
  int actualSpeed = abs(readActualSpeedVSD[0]);                                                                     // Update all of the neccessary objects on the display with values from the VSD modbus reads and the user speed input slider.
  Serial.println( statusword);
  delay(10);
  Serial.println( voltage);
  delay(10);
  Serial.println( current);
  delay(10);
  Serial.println( frequency);
  delay(10);
  Serial.println( actualSpeed);

 delay(100);
 int BUTTONSTATE = digitalRead(StartButton);
 delay(100);
 {
 Serial.println(BUTTONSTATE);
 if(BUTTONSTATE==1)
 {
 writeControlWordVSD[0]=15;
 Serial.println("Writing Control word = 15");
 }
 delay(10);
 modbus_update();
 delay(1000);
 }
}
[ATV312_comm_var_2009_eng.pdf](https://github.com/angeloc/simplemodbusng/files/2592020/ATV312_comm_var_2009_eng.pdf)

limitedAtonement, 2020-08-09:
I'm guessing this can be closed. If it's still an issue, where is `packetPointer` defined?

closed 2020-11-10 by angeloc