angeloc/simplemodbusng #24: Compiling with esp8266

eevora, 2019-11-08:
I´ve been using this library with arduino mega, it works very good, but now I want to use it with esp 8266 and the arduino IDE gives me an error ¨error compiling for esp 8266¨, this is because this library does not work with esp8266??

angeloc, 2019-11-08:
I don't really know, never tried that! For sure it has some dependencies on arduino internals, so it probably should be adapted for esp8266.
Anyway, what's the complete output error log?

eevora, 2019-11-08:
The Error Log is this:
Arduino:1.8.10 (Windows 10), Tarjeta:"Generic ESP8266 Module, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), ck, 26 MHz, 40MHz, DOUT (compatible), 512K (no SPIFFS), 2, nonos-sdk 2.2.1 (legacy), v2 Lower Memory, Disabled, None, Only Sketch, 9600"

C:\Users\Ernest\Documents\Arduino\libraries\SimpleModbusMasterV2rev2\SimpleModbusMaster.cpp: In function 'void modbus_configure(HardwareSerial*, long int, unsigned char, long int, long int, unsigned char, unsigned char, Packet*, unsigned int, unsigned int*)':

C:\Users\Ernest\Documents\Arduino\libraries\SimpleModbusMasterV2rev2\SimpleModbusMaster.cpp:450:38: error: invalid conversion from 'unsigned char' to 'SerialConfig' [-fpermissive]

  (*ModbusPort).begin(baud, byteFormat);

                                      ^

In file included from C:\Users\Ernest\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Arduino.h:263:0,

                 from C:\Users\Ernest\Documents\Arduino\libraries\SimpleModbusMasterV2rev2\SimpleModbusMaster.h:83,

                 from C:\Users\Ernest\Documents\Arduino\libraries\SimpleModbusMasterV2rev2\SimpleModbusMaster.cpp:1:

C:\Users\Ernest\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h:78:10: error:   initializing argument 2 of 'void HardwareSerial::begin(long unsigned int, SerialConfig)' [-fpermissive]

     void begin(unsigned long baud, SerialConfig config)

          ^

Se encontraron varias bibliotecas para "SimpleModbusMaster.h"
Usado: C:\Users\Ernest\Documents\Arduino\libraries\SimpleModbusMasterV2rev2
Se encontraron varias bibliotecas para "SPI.h"
Usado: C:\Users\Ernest\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SPI
Se encontraron varias bibliotecas para "SD.h"
Usado: C:\Users\Ernest\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SD
 No usado: C:\Users\Ernest\Downloads\arduino-1.8.10-windows\arduino-1.8.10\libraries\SD
Se encontraron varias bibliotecas para "SDFS.h"
Usado: C:\Users\Ernest\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SDFS
Se encontraron varias bibliotecas para "SdFat.h"
Usado: C:\Users\Ernest\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266SdFat
exit status 1
Error compilando para la tarjeta Generic ESP8266 Module.

Este informe podría contener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.
Maybe its something with the serial port of ESP8266 that its not compatible with the serial of the arduino mega, sorry the log its in spanish I´m from Cuba

paccerdk, 2020-01-17:
Using the library on esp8266 without any problems. (vscode with platformio, but should be same with arduino ide)
Check your setup(s) and perhaps test with a blink sketch or similar to check that normal compilation is working.

closed 2020-11-10 by angeloc