angeloc/simplemodbusng #26: wiring with arduino as slave and by RS485 to master?

bigplik, 2020-10-20:
hi, I appreciate if someone could explain wiring arduino with other device (master) by RS485 or RS232, when using code from this library?

I understand code provided here can act as slave for some modbus rtu master simulation on PC, but what about using it with other devices (PLC, HMI, another arduino)?

angeloc, 2020-11-10:
Could you explain your setup? RS485 is a pretty simple bus, only 2 wires, if you correctly wire them you can talk with master or slaves

ashrafulkhn, 2023-12-24:
In case you are trying to connect to a PLC or a HMI then connect the A and B pin of the PLC to the RS485 converter module you have used with Arduino. Make sure that A pin should be connected to A and B pin should be connected to B pin respectively (Not like tx->Rx and rx->tx).

Natcat-kode, 2025-09-07:
Hi I am struggling to install this library, it keeps returning an erroir message.

closed 2025-10-08 by angeloc

ashrafulkhn, 2025-11-09:
Can you please share the error message. A screenshot or the log should be helpful enough. Can you also tell the platform details where you are trying to install.