angeloc/simplemodbusng #37: Arduino Fio Compatible

OptimalControls, 2021-12-01:
Dear people, I tested this library under Arduino Fio, unfortunetely, if you have more than 4 slaves, they fail to comunicate. 
I tested the same with Nano, without any problems, but Fio seems to need further adjusments in the library, maybe because of its 8mhz clock. Please, if you have any update or work around let me know.
Regards,
Oscar

angeloc, 2021-12-01:
I cannot replicate your setup, it would be more than welcome if you can do a proper debug to understand where the problem lies.
Of course the master is subject to the cpu speed and bus speed, if your cpu is running slowly and the bus is not working at full speed, you cannot have enough time to interrogate all the nodes.
Usually the master is a bigger and heavire machine like a linux embedded.

negroKiordi, 2022-06-28:
> Dear people, I tested this library under Arduino Fio, unfortunetely, if you have more than 4 slaves, they fail to comunicate. I tested the same with Nano, without any problems, but Fio seems to need [...]

Hello, may you have the same problem as me. I have open the issue 39 "". I have found a solution changing the implementation of the time out T1_5 in modbus_update() function.